StateSmith/StateSmith #381: simulator - improve guard evaluation dialog

adamfk, 2024-08-18:
Two main problems.

# No context about guards
There's no context about what state behavior is being evaluated. You just get prompted to evaluate some code.

[image]

We should include state name and state behavior as UML.

# Dialog is a bit confusing
OK for true and CANCEL for false isn't as intuitive

adamfk, 2024-08-18:
I tried using https://sweetalert.js.org/guides/#advanced-examples, but it returns a promise. Will need to invest in async features for simulator first I believe.

adamfk, 2024-08-18:
After change 
[image]

closed 2024-08-19 by adamfk

krzysztofkuczek, 2025-09-22:
Hi,
I can see, that the problem is fixed, but today using ss.cli 0.19 got same behavior.
What should i do.
Thank you in advance.

[image]

adamfk, 2025-09-22:
Hi @krzysztofkuczek,

This dialog box is normal for evaluating guard expressions. It will be improved further in the future, but is better than it was.

Does pressing "OK" allow the transition to be taken?

krzysztofkuczek, 2025-09-22:
Hi,
Yes it works, but in fact, the real expression is not taken under consideration....

adamfk, 2025-09-22:
> Hi, Yes it works, but in fact, the real expression is not taken under consideration....

That is true. Because StateSmith lets user code contain anything, the simulator doesn't try to evaluate expressions. That's why it has to ask the user with the dialog box.

One day, StateSmith simulator will be smart enough to understand common expressions, but probably not for a few years.

Can you please open a new issue to improve simulator documentation to make this more clear? I'm currently traveling.

commit 3e8d912e (2024-08-18): simulator - improve guard evaluation dialog

Closes #381