StateSmith/StateSmith #385: C Add CFileBottom / HFileBottom

timscheuermann, 2024-08-28:
Is it possible to add a CFileBottom to the last line of the generate source/header file in C?
It would be very nice, because I need to remove the complete sourcecode with an makro if and an macro else at the end to exclude the source file.
And of course it would have other use cases.

Thanks.

adamfk, 2024-08-28:
I've been thinking about the same thing recently. I'd be happy to implement this in a week or two, or I can help you implement it if you are interested. It should be pretty easy with a bit of guidance.

Note: we are also adding support for standard `#ifdef` include guards: https://github.com/StateSmith/StateSmith/pull/383

Will merge the above PR soon. Test builds are available though: `dotnet tool install -g statesmith.cli --version 0.11.1-test-1`

adamfk, 2024-08-28:
I had time on my lunch break to code this in. Will be released soon.

adamfk, 2024-08-28:
Thanks for the suggestion @timscheuermann :)

closed 2024-08-29 by adamfk

adamfk, 2024-08-29:
released in lib version: 0.12.2-alpha
released in cli version: 0.11.2

timscheuermann, 2024-08-30:
Another question: How do I install the correct version?
When I run this, the CFileBottom/HFileBottom seems to be missing.
`dotnet tool install --global StateSmith.Cli --allow-downgrade --version 0.11.2 -v d
[NuGet Manager] [Info]   GET https://api.nuget.org/v3/registration5-gz-semver2/statesmith.cli/index.json
[NuGet Manager] [Info]   OK https://api.nuget.org/v3/registration5-gz-semver2/statesmith.cli/index.json 237 ms
[NuGet Manager] [Info]   GET https://api.nuget.org/v3-flatcontainer/statesmith.cli/index.json
[NuGet Manager] [Info]   OK https://api.nuget.org/v3-flatcontainer/statesmith.cli/index.json 120 ms
[NuGet Manager] [Info]   GET https://api.nuget.org/v3-flatcontainer/statesmith.cli/0.11.2/statesmith.cli.0.11.2.nupkg
[NuGet Manager] [Info]   OK https://api.nuget.org/v3-flatcontainer/statesmith.cli/0.11.2/statesmith.cli.0.11.2.nupkg 24 ms
Die Überprüfung der NuGet-Paketsignatur wird übersprungen.`

adamfk, 2024-08-30:
How are you specifying your settings? With toml in the diagram?

Here is the settings documentation:
https://github.com/StateSmith/StateSmith/blob/main/docs/settings.md#renderconfigchfilebottom

If you are using the older diagram based render config, I'm no longer adding new settings to it. Use toml instead: https://github.com/StateSmith/StateSmith/wiki/Diagram-Based-Render-Config

timscheuermann, 2024-09-03:
I got it working, I waited until 0.12.0, now it works directly via toml.
Thank you for adding the feature so quickly.

adamfk, 2024-09-03:
You're welcome! I hope it works well for you.