StateSmith/StateSmith #420: Allow running cli tool non-interactively

ptensschi, 2024-11-04:
Currently when the cli-tool is ran for the first time, and no `tool-settings.json` is there, it would show the _Welcome! Let's set up your tool settings_ interaction. This is somewhat cumbersome in continuous integration and container development use cases, where I just want to do the conversion at the first call of the tool and leave any updating to the container image.

Please provide a way to use the tool "config-file-less", e.g. using suitable command line options, so that it just does the conversion and nothing else.

Btw, this is such a great tool, keep up the awesome work!

adamfk, 2024-11-04:
Hi @ptensschi! Good call. I've been meaning to get to this. I'll have something ready in a few days.

The option could be something like `--no-ask`? Then it wouldn't ask if you wanted to setup, or check for updates. 
```
ss.cli run -hr --no-ask
```

Thoughts?

Do you mind sharing a bit about how you are using StateSmith? Programming language? What you are using a state machine for (embedded, game dev, UI, ???). No worries if you can't share. It's just really cool to see how it is being used :)

closed 2024-11-06 by adamfk

adamfk, 2024-11-07:
CLI tests failed on github server due to latest node.js version (23) printing extra warning (not due to our code). Here's the fix: https://github.com/StateSmith/StateSmith/commit/b6eae1c93951bb3fec4ba4b492cc01b19b56ae56

ptensschi, 2024-12-16:
Hi Adam, thanks heaps for fixing this, works like a charm!

As for the use of StateSmith, I use this in embedded mostly (pressing buttons, running and braking motors, lighting leds, starting and stopping other actuators).

The interactive simulator it helpful here in particular, because you can "feel" it before throwing it onto real embedded hardware.

adamfk, 2024-12-17:
Very cool @ptensschi. Thanks for sharing :)

commit 1187ff48 (2024-11-06): cli - Add `--no-ask` option to `run` verb. Close #420