StateSmith/StateSmith #421: Feature Request: Hash Table Optimization for Event Handling

DGYin, 2024-11-08:
**Background**
I am currently using StateSmith to develop state machines on low-power embedded platforms, specifically an STM32F103 microcontroller with a 72MHz clock speed.
The xxxSm_dispatch_event function utilizes a switch statement to handle events. When dealing with a large number of events and a high polling frequency, such as a state machine with over a hundred events or states being polled at 500Hz, the current implementation can become a significant demand on CPU resources. This has led to resource starvation for other threads running on the same microcontroller.

**Request**
To address this issue, I would like to request the addition of an option during code generation that allows the use of a hash table to optimize event handling functions. The goal is to reduce the computational overhead associated with event dispatching, especially in scenarios where there are many events and a high polling rate.
The idea is to have a configuration option within StateSmith that, when enabled, generates event handling code using a hash table instead of switch statements. The hash table would map event IDs to their corresponding handler functions, allowing for more efficient event processing.

**More**
I used to write my own switch statements for very simple state machines, but it was neither intuitive nor easy to maintain, often leading to confusion. Since adopting StateSmith, my productivity has significantly increased, and I am truly grateful for the hard work and dedication of your team. It would be a great honor and a pleasure for me if you could consider my suggestion, as it would further enhance the efficiency and performance of StateSmith in resource-constrained environments.
Due to my limited English proficiency, I have had this message polished by an AI. I hope for your understanding and consideration :)

adamfk, 2024-11-08:
Hello @DGYin. Interesting suggestion! I've never seen a design with a hundred events! Interesting! Is this for a parser or something? We should be able to find a good solution to your problem.

What programming language are you using? Most languages implement switch statements fairly efficiently. They compute an offset into a table and then jump into that table. Python's match statements are more like a bunch of if, else if, else if statements and much lower performance.

Do you have optimizations enabled?

# Switch Statements
If you are using C/C++ and have optimizations enabled, a switch statement over the state IDs is usually very fast. About 6 assembly instructions on a Cortex M3. See here: https://gcc.godbolt.org/z/ThGe71MKb

Without optimizations enabled, it will take about 12 assembly instructions. See here: https://gcc.godbolt.org/z/qYKbsbrfM

It will then take another 6 or 12 assembly instructions for the event switch.

In total, the event dispatch function will take around 12 to 24 assembly instructions. At 72 MHz, this should complete very quickly. Have you tried measuring the dispatch function time or looking at the assembly?

# State & Event Table
One potential solution is a constant table of function pointers. Like a 2 dimensional array. 1 dimension is for the state ID, the other dimension is for the event ID. This approach can waste a fair amount of ROM though if you have many events that are ignored my most states.

# Hash table
I'll think about the hash table suggestion too. My immediate thought is that it will likely require much more CPU time than switch statements or an event table unless StateSmith searches for a perfect hashing algorithm.

adamfk, 2025-01-19:
@DGYin I'm closing for now, but we can re-open if you have time to respond.

closed 2025-01-19 by adamfk