StateSmith/StateSmith #424: entryPoint not working with events

sdaglish, 2024-11-19:
I'm having issue with multiple entry points, when the transition to an entry point is an event `EVENT`, instead of a check `[EVENT = true]`

If I run the following state diagram:

```
state PowerOnReason

[*] --> PowerOnReason

state IDLE {
    state IdleToDisplayOff <<entryPoint>>
    state IdleToDisplayOn <<entryPoint>>

    state IDLE_DISPLAY_OFF {
    }

    state IDLE_DISPLAY_ON {
    }
    IdleToDisplayOff  --> IDLE_DISPLAY_OFF
    IdleToDisplayOn --> IDLE_DISPLAY_ON
}

PowerOnReason --> IdleToDisplayOn : [REASON1 == true]
PowerOnReason --> IdleToDisplayOff : [REASON2 == true]
```
SS generates the source fine.

However, if I change the transitions to the entry points to

```
PowerOnReason --> IdleToDisplayOn : REASON1
PowerOnReason --> IdleToDisplayOff : REASON2
```

I get the following error:

```
Failed parsing diagram edge
from: ROOT.PowerOnReason
to:   ROOT.IDLE
Edge label: `REASON1via entry IdleToDisplayOn`
Reason(s): no viable alternative at input ' entry' at line 1 column 11. Offending symbol: `entry`.
           mismatched input 'entry' expecting <EOF> at line 1 column 11. Offending symbol: `entry`.
Edge diagram id: line_42_column_0

>>>> RELATED HELP <<<<
https://github.com/StateSmith/StateSmith/issues/174

Related error info/debug settings: 'dumpErrorsToFile', 'propagateExceptions'. See https://github.com/StateSmith/StateSmith/blob/main/docs/settings.md .
StateSmith.Runner.FinishedWithFailureException: Exception of type 'StateSmith.Runner.FinishedWithFailureException' was thrown.
   at StateSmith.Runner.SmRunner.PrintAndThrowIfPreDiagramSettingsException() in C:\Users\akruck\Documents\code\ss\StateSmith\src\StateSmith\Runner\SmRunner.cs:line 140
   at StateSmith.Runner.SmRunner.Run() in C:\Users\akruck\Documents\code\ss\StateSmith\src\StateSmith\Runner\SmRunner.cs:line 107
   at Submission#0.<<Initialize>>d__0.MoveNext() in /home/steven/test/tset.csx:line 12
--- End of stack trace from previous location ---
   at Dotnet.Script.Core.ScriptRunner.Execute[TReturn](String dllPath, IEnumerable`1 commandLineArgs) in C:\Users\runneradmin\AppData\Local\Temp\tmp8BAF\Dotnet.Script.Core\ScriptRunner.cs:line 110
```

adamfk, 2024-11-20:
Hi @sdaglish. The fix for this will be released in a few days.

In the meantime, you should be able to workaround the problem by adding a dummy guard `[true]` after the event like:

```diff
-PowerOnReason --> IdleToDisplayOn : REASON1
+PowerOnReason --> IdleToDisplayOn : REASON1 [true]
```

adamfk, 2025-01-08:
Will be available in next release. Hopefully in a few days.

closed 2025-01-08 by adamfk

adamfk, 2025-01-08:
Should be fixed now in cli 0.17.4. Would be great if you could confirm. Thanks!

sdaglish, 2025-04-07:
Sorry for delay. Can confirm this fixes my issue.
Thanks.

commit f2c0413e (2025-01-08): lib - fix #424 PlantUML entry/exit point conversion failure.