StateSmith/StateSmith #425: DrawIO: New Line in Transition

timscheuermann, 2024-11-27:
When I add a transition with an Event and a condition from one state to another, i get different code generation results when i add a new line between the Event and the condition.
For example when adding the transition
EVENT_BUTTON_START_UPDATE_CLICKED [m_form.IEB900_ValidFileSelected() >= 0] in only one line, the event and condition both gets inserted. 
But when I do a linebreak the event is ignored and statesmith inserts a DO-Event instead, which seems wrong although nothing really has changed in the diagram except for the new line. This transition looks like this:
EVENT_BUTTON_START_UPDATE_CLICKED 
[m_form.IEB900_ValidFileSelected() >= 0]

As a workaround I insert the opening square bracket in the first line and the rest of the condition in the second line, which gives the expected result:
EVENT_BUTTON_START_UPDATE_CLICKED [
m_form.IEB900_ValidFileSelected() >= 0]

Is this a bug or is there a better workaround? Because me as a user would not expect this behaviour.

adamfk, 2024-11-27:
I think what's happening when there's a line break in between is that
StateSmith sees them as two separate Behaviors sharing the same edge/arrow.
One behavior is just your event, and the other behavior is just the guard.

That's a feature that might need some changes.

I'll do more investigation on my break/lunch today.

Thanks for raising this issue.

timscheuermann, 2024-11-28:
First of all, I do not have big experience in Statesmith's behaviour on the different pattern.
My thought is to leave the behaviour as it is and adding a setting which allows the user to switch between multiple possibilities. The default setting would be the current behaviour. This means that all current diagrams will result in the same expected code. If the user wants other behaviour, he just changes it by changing the setting and its on him.

adamfk, 2024-11-29:
This could work. It would also be nice to have a standard syntax though. Then I can look at any StateSmith diagram and not have to worry about the settings.

I'll scan through all the examples this weekend and see if I come across any places that would break. It might not actually affect anyone.