StateSmith/StateSmith #427: BaseClassCode (unexpectedly?) applied to Vars in Cpp Transpiler

ElFishle, 2024-12-12:
I'm playing around with the C++ transpiler and tried to get some inheritance pattern to work. It basically plays well, but behaves somewhar unexpected. I use this as my RenderConfig.Cpp:
```
[RenderConfig]
AutoExpandedVars = """
    int count;
"""
[RenderConfig.Cpp]
BaseClassCode = "public LightController"
[SmRunnerSettings]
transpilerId = "Cpp"
```
This results expectedly in:
```
[...]
// Generated state machine
class LightSm : public LightController
{
public:
    enum class EventId: uint8_t
    {
[...]
```

But also in:
```
[...]
    // State machine variables. Can be used for inputs, outputs, user variables...
    class Vars : public LightController
    {
    public:
        int count;
    };
[...]
```

Is this intended that the BaseClassCode is attached to the contained `Vars` class? Or did I get something else wrong? Otherwise I would expect that the `Vars` class should not receive the "BaseClassCode".

Why that hits me: I'm using something like this as base class to be able to directly call the start method:
```
class LightController
{
public:
	virtual void start() = 0;
[...]
};
```

adamfk, 2024-12-12:
Thanks for the report! I'll fix soon, but in the mean time, you can simply move any vars from the state machine to your base class. See the `AutoExpandedVars` in your toml config. Then no Vars class will be generated.

ElFishle, 2024-12-12:
Thank you for pointing out the workaround and also in advance for fixing.

adamfk, 2025-01-08:
Will be available in next release. Might be a few days.

closed 2025-01-08 by adamfk

adamfk, 2025-01-08:
Should be fixed now in cli 0.17.4. Would be great if you could confirm. Thanks!

ElFishle, 2025-01-08:
Confirmed with ss.cli-win-x64-compressed.exe from [Release 0.17.4](https://github.com/StateSmith/StateSmith/releases/tag/cli-v0.17.4).

Now I get:
```
[...]
// Generated state machine
class LightSm : public LightController
{
public:
    enum class EventId: uint8_t
    {
[...]
```

and also:
```
[...]
    // State machine variables. Can be used for inputs, outputs, user variables...
    class Vars
    {
    public:
        int count;
    };
[...]

```
Thank you very much for fixing and your great project.

adamfk, 2025-01-08:
Thanks for confirming :)

commit 04afa9ae (2025-01-08): Fix Cpp transpiler so that nested `Vars` class doesn't extend user base class. Closes #427