StateSmith/StateSmith #428: Allow use of .toml file instead of embedded toml on cli

ptensschi, 2024-12-16:
The cli is a game changer for CI, because the csx solution always tries to load dependencies at runtime. 
However, in case you have multiple multiple projects all sharing the same settings like "IRenderConfigCpp", it is cumbersome (and violating the DRY rule) to embed the toml into every diagram.
It would be great having the possiblity to reference a toml file on the command line ( e.g. `statesmith run --with-toml my_shared_settings.toml --no-csx --here...`) as an alternative way to using .csx file.

adamfk, 2024-12-17:
I like this idea. I'm thinking the external toml file will be applied first and then any toml settings in the diagram will be applied.

adamfk, 2025-02-26:
I've been thinking more about this lately. I think it would be useful to allow multiple toml files.

one of my use cases is in using the same state machine diagram for multiple programming languages. Will try to implement soon. Shouldn't be too bad.

Example: `ss.cli run --with-toml shared1.toml --with-toml shared2.toml --no-csx --here`

adamfk, 2025-05-17:
Hi @ptensschi. Sorry for the long delay!

Now that the [Embedded Online Conference](https://embeddedonlineconference.com/session/Visualize_Your_State_Machines) is over (it was really good), I have more time to finish this up.

Working through implementation details now.

## Rename to `--config-files`
Q1. I'd like to use a slightly different CLI argument name I think: `--config-files` rather than `--with-toml`. This allows other future config formats if we want like yaml/json.

Ex: `statesmith run --config-files my_shared_settings.toml --no-csx --here`

## Relative Paths?
Q2. Should the config file toml paths be relative to current terminal directory or each found diagram file?

I think it makes sense to use current terminal directory.

Ex: `ss.cli run ButtonSm.plantuml --config-files ../shared.toml ButtonSm.toml`. Uses [new simplified run args](https://github.com/StateSmith/StateSmith/issues/422).

## External config files from toml settings
I would like to add support for users to have almost no toml config inside their diagrams.

For example, a `ButtonSm.plantuml` file make look like this:
```
@startuml ButtonSm

state RELEASED_DEBOUNCE
state RELEASED_STABLE
state PRESSED_DEBOUNCE
[*] -> RELEASED_DEBOUNCE
' ...

/'! $CONFIG : toml
# NOTE! Files paths below are relative to diagram!!!
SmRunnerSettings.configFiles = [ "../base.toml", "ButtonSm.toml" ]
'/
@enduml
```
The toml inside of a plantuml comment works, but is kinda hard to follow sometimes.

Then we could have `ButtonSm.toml` that looks just like this (with ide highlighting):
```toml
########################### StateSmith config ###########################
# More info: https://github.com/StateSmith/StateSmith/issues/335

SmRunnerSettings.transpilerId = "C99"

RenderConfig.C.HFileUseExternC = true

RenderConfig.C.HFileIncludes = """
    #include <stdint.h> // for fixed width integer state machine variables below
    """

RenderConfig.AutoExpandedVars = """
    // ----- INPUT VARIABLES -----
    uint16_t timer_ms;
    uint8_t input_active: 1;

    // ----- OUTPUT VARIABLES -----
    uint8_t release_event: 1;
    uint8_t press_event: 1;
    uint8_t long_event: 1;
    uint8_t press_status: 1;
    uint8_t long_status: 1;
    """

# Declare generated enumerations as "packed" for GCC. Often saves some RAM/flash. Not needed for C++11 transpiler.
RenderConfig.C.CEnumDeclarer = "typedef enum __attribute__((packed)) {enumName}"
```

I'll have to put in some kind of check to prevent infinite loops (toml files that reference each other somehow), but that shouldn't be too bad. I might just be lazy for now and limit it to 50 tries :P

Q3. Thoughts?