StateSmith/StateSmith #429: Python generated class doesn't call the base class constructor.

alexis-boisserand, 2025-01-06:
In Python, if a constructor is declared, it has to explicitly call the base class constructor, otherwise it's never called.
Currently the generated class constructor looks something like this:
`python
    # State machine constructor. Must be called before start or dispatch event functions. Not thread safe.
    def __init__(self):
        # Used internally by state machine. Feel free to inspect, but don't modify.
        self.stateId = None
`
It could be rewritten like this, so that all potential arguments get passed to the base class constructor:
`python
    # State machine constructor. Must be called before start or dispatch event functions. Not thread safe.
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        # Used internally by state machine. Feel free to inspect, but don't modify.
        self.stateId = None
`

Here is my attempt to implement it that way: 
https://github.com/StateSmith/StateSmith/compare/main...alexis-boisserand:StateSmith:main

adamfk, 2025-01-06:
Thanks @alexis-boisserand ! Can you open a PR from your fork?

adamfk, 2025-01-07:
No worries! I can validate and tidy up anything that comes up. Thanks for the fix :)

closed 2025-01-08 by adamfk

adamfk, 2025-01-08:
Should be fixed now in cli 0.17.4. Would be great if you could confirm. Thanks!

alexis-boisserand, 2025-01-08:
I updated ss.cli to 0.17.4 and yes, I confirm it now works as expected.

adamfk, 2025-01-08:
Thanks for confirming :)

commit 56e18de9 (2025-01-08): Merge pull request #430 from alexis-boisserand/main

Calls base class constructor when generating python code. Fixes #429