StateSmith/StateSmith #432: comments generate too much

WXrock, 2025-01-13:
In the automatically generated C/C++ code, there are too many comments. Could a switch be added to control whether comments are generated?

adamfk, 2025-01-14:
I think the comments are great while learning about the generated code and
algorithm.

But after you understand that, they are very verbose. I'd like to see this
feature added as well.

I think it would be helpful to have different comment verbosity levels. For
example, we likely want to keep function comments about thread safety, but
not comments explaining state behaviors.

Would you be able to help implement this? I can provide guidance.

WXrock, 2025-01-15:
I'm not so good at C# progming, but maybe I can try it

adamfk, 2025-01-15:
Before we start implementing, can you provide examples of the kinds of messages that you think we should always keep and which ones can be silenced?

I'm thinking the comment verbosity level should be a setting in the TOML settings. Do you agree? https://github.com/StateSmith/StateSmith/blob/main/docs/settings.md

Once we have the design idea figured out, I'll start a branch that provides the new comment verbosity setting. We use dependency injection which feels kinda magical if you haven't used it in C# before. So I'll do that part.

WXrock, 2025-01-16:
I agree with the TOML settings idea, I think maybe the comments should be divided into three levels: 
**NONE**, **INFO** and **DEBUG**.
For **NONE** level, we can silence all of them and keep the code as clean as possible.
For **INFO** level, we can keep the function comments and struct comments such as:

// State machine constructor. Must be called before start or dispatch event functions. Not thread safe.
// Starts the state machine. Must be called before dispatching events. Not thread safe.
// Dispatches an event to the state machine. Not thread safe.
// Used internally by state machine. Feel free to inspect, but don't modify.

For **DEBUG** level, we can keep all the comments that explains how statesmith works which is helpful for Debugging.

adamfk, 2025-01-22:
Had to travel for work. Might be another week or so until I can get you started in this. Sorry for the delay. It's a good idea.

adamfk, 2025-02-16:
Sorry for the delay. Here's a status message I posted on StateSmith discord:

Hi all, sorry for the slow pace of development recently. Especially in releasing after PRs. I started a new job in January at a really cool company launching hundreds of Low Earth Orbit satellites. That's been exciting, but busy. On top of that I've been sick for a month too. Hopefully better soon. Just wanted to let people know I'm not intending to stop StateSmith development. Just physically not really capable right now. Hopefully a new release soon with emmby and wxrock contributions.

WXrock, 2025-02-19:
Sincerely wish you a speedy recovery.