StateSmith/StateSmith #434: [simulator] LOG info disappear when there's too many events in FSM

WXrock, 2025-01-17:
[image]

adamfk, 2025-01-18:
Hmm. We will have to fix this. Hopefully we can just add a scroll bar if there are a lot of events.

adamfk, 2025-01-31:
Do you mind if I ask what you are working on? How many events do you have?

You might be able to combine more generic events with guard conditions like `RECEIVE [rx_char == 'a']` and `RECEIVE [rx_char == 'b']` instead of events like `RECEIVE_A` and `RECEIVE_B`.

I'll see if there is a simple solution for this. Hopefully just setting a scroll bar option in the web page CSS.

WXrock, 2025-02-01:
Hi @adamfk thank you for your advice，I am developing a robotics project which consists of more than 5 hierarchical state machines and more than 20 events. I used draw.io to draw them together and simulate them.