StateSmith/StateSmith #440: StateId-enum custom order of numbers

timscheuermann, 2025-02-13:
Hi, 
is there a possibility to change the orders of the State-Numbers inside the StateId-enum?

I send the state of my first system as an integer to another system to get an info about the active state. But when I add or remove some state in the Statemachine, the StateId-number could have changed also and i could accidentally mix up the states.
If this is not possible, i would have to store all stateId enums inside the second system and ask for a version of the running statemachine, which would be also okay. But have you thought about this?
Thank you.

adamfk, 2025-02-13:
Interesting. I don't think I've ever considered this. I can see how it
would be useful, but don't have an automated way to achieve this right now.

When I've needed to give the status of one state machine to another (or a
UI) I've usually manually set a state machine variable with the status.
This usually works out alright because I often have more states than status
values.

For example, I might have a custom STARTING status that is used for 5 or so
states.

I set the status manually when entering or exiting states.

Does that help at all?

If we were to add this feature, how would you see it working for the user?
How and where should they set it? Might not be too hard.

timscheuermann, 2025-02-14:
> When I've needed to give the status of one state machine to another (or a
UI) I've usually manually set a state machine variable with the status.
This usually works out alright because I often have more states than status
values.
For example, I might have a custom STARTING status that is used for 5 or so
states.
I set the status manually when entering or exiting states.
Does that help at all?

The problem is that i could have made changes to the statemachine while the other system depends on the state-ids but has an older enum.

> If we were to add this feature, how would you see it working for the user? How and where should they set it? Might not be too hard.

First thought: Maybe we call a manual sorting function when running dotnet sm.csx which lists all states and the user can assign a new order to it?

I dont know if this feature will be worth to get implemented but that is not up to me.
For my case I will just go the secure way at the moment and store every stateId-enum for every version i have done in the past.

adamfk, 2025-02-14:
If I understand correctly, I think you could use static assertions to keep
a custom enum in synch with your state machine.

static assert against generated state enum values and size

https://github.com/StateSmith/tutorial-3/blob/a2c9f494cd62247b9bed34e6efb42cbbc11ab95e/lesson-4/c99-gcc/LightSm.h#L24-L37

I've done this before too. It catches anything out of synch at compile time
and no additional storage.

If the enum value is stored in a file or something this might help:
https://github.com/StateSmith/StateSmith-examples/blob/main/history-storage-1/README.md#dont-rely-on-autogenerated-history-id

I'm open to us implementing this as I do see the usefulness but I'd like to
prototype it with the .CSX file approach. It allows your custom .CSX file
to hook into StateSmith and customize it fairly easily while testing it
against your application code.

Any interest in prototyping it with my help?

timscheuermann, 2025-02-19:
>I'm open to us implementing this as I do see the usefulness but I'd like to
prototype it with the .CSX file approach. It allows your custom .CSX file
to hook into StateSmith and customize it fairly easily while testing it
against your application code.
Any interest in prototyping it with my help?

Do you mean the approach of manually editing the enum? We should definitely give it a try. How do we start?