StateSmith/StateSmith #441: Use C++14 or later in Code Generation

emrainey, 2025-02-13:
The Generated C++ code could use with some minor improvements which would make it more compatible with more recent standards, even under MISRA rules. 

* `enum class` instead of using a packed enum
* (customizable) namespace to separate all State related types out of the global namespace. 
* Pure Virtual Interfaces for "callback" types. 
* References instead of pointers
* `using` instead of `typedef`

adamfk, 2025-02-14:
I probably need to do a better job of advertising this, but StateSmith has
had more proper C++11 support for a while.

Check out:
https://github.com/StateSmith/tutorial-3/blob/main/lesson-4/cpp2-gcc/LightSm.h

and settings for CPP for namespace.
https://github.com/StateSmith/StateSmith/blob/main/docs/settings.md#renderconfigcpp

These newer diagrams use a switch/case style algorithm (not function
pointers).

MISRA will probably not like the multiple return points though in some
state handlers.

Does this help?

emrainey, 2025-02-18:
It does! It seems like, to my uneducated eye, that there could be some improvement by passing the events into the handlers as parameters instead of have a handler per event? Perhaps it just moves _where_ the case statement / if else code is instead of eliminating it. 

`enter(State::XXXXX);` instead of `XXXXX_enter();`
or
`process(State::X, Event::Y);` .

As I think about it, it makes the API cleaner but I auppose makes it harder to know if you've implemented all the conditions? Was this intentional in the code generation?

emrainey, 2025-02-18:
By having the class extend the base which has the callbacks, the reference to the callbacks is eliminated but now the constructor of the base class can't be anything but the default?

adamfk, 2025-02-18:
> It does! It seems like, to my uneducated eye, that there could be some improvement by passing the events into the handlers as parameters instead of have a handler per event? Perhaps it just moves _where_ the case statement / if else code is instead of eliminating it.
> 
> `enter(State::XXXXX);` instead of `XXXXX_enter();` or `process(State::X, Event::Y);` .
> 
> As I think about it, it makes the API cleaner but I suppose makes it harder to know if you've implemented all the conditions? Was this intentional in the code generation?

StateSmith currently supports 2 algorithms (for some languages): https://github.com/StateSmith/StateSmith/wiki/Algorithms

The intent is to keep adding algorithms over time. The next one we add for C++ will probably prioritize speed. It would be a 2d array of state event function pointers for maximum speed. 1 array dimension for state id, 1 array dimension for event id. Array contains function pointers.

By defining the state event handlers like we currently do, we can support a number of different algorithms (switch/case, 2d array, dynamic array, ...) across multiple programming languages without too much work. The main difference between those algorithms is just how you go from state id and event id to function. StateSmith is flexible enough to support fully custom algorithms for just specific languages too if you'd like to write one. It's a fair amount of work though :)

I modeled the existing switch/case statements based on what most people seem to handwrite. I wanted it to look rather simple, approachable and still efficient.

There will be some changes to the generated internal code when we want to support one of the last big features (orthogonal states): https://github.com/StateSmith/StateSmith/issues/75 The generated event handlers will likely need to either return a status or take an event context: https://github.com/StateSmith/StateSmith/issues/183

It isn't perfect yet, but it is still very useful as long you don't have extreme speed needs. Definitely open to ideas and suggestions though :)

adamfk, 2025-02-18:
> By having the class extend the base which has the callbacks, the reference to the callbacks is eliminated but now the constructor of the base class can't be anything but the default?

That is currently a limitation. My current workaround is that I indeed have a default constructor for the base class, but if it needs references to something else, I use std::reference_wrapper so that I can change it after construction.

Composition works well (state machine has a reference to another class), but then you have to use that referenced object like `some_object.do_stuf()` instead of just `do_stuff()`. This clutters the diagram up a bit. I often shorten the reference to just a single letter though.

I've been meaning to open a ticket to allow defining custom state machine constructors. I think this flexibility would help a lot. If the user provides custom constructor code, then StateSmith wouldn't generate a blank ctor like it does now. https://github.com/StateSmith/StateSmith/issues/443

Would that be useful for you?

emrainey, 2025-02-18:
Other State Machines systems (Yakindu) I've used simply have the StateMachine as an object which takes a pure virtual interface reference at construction. This way it's mockable in GTest or others.  It is then up to the dev to implement that object which extends the interface. The testablility of interfaces over base class extention I think makes them at least my preferred way.

adamfk, 2025-02-23:
> Other State Machines systems (Yakindu) I've used simply have the StateMachine as an object which takes a pure virtual interface reference at construction. This way it's mockable in GTest or others. [...]

Can you take a look at this example? https://github.com/StateSmith/StateSmith-examples/tree/main/cpp-inheritance-1

It shows how you can test a state machine fairly easily via the base class "interface".

You can also defined the "interface" directly in the generated state machine if you want. See this example: https://github.com/StateSmith/StateSmith-examples/tree/main/cpp-inheritance-2

Thoughts on the above?

adamfk, 2025-02-26:
I think this issue is OK to close now.

If there's something in specific that we should improve, let me know!

closed 2025-02-26 by adamfk