StateSmith/StateSmith #450: Add customizable base class for C++ statemachines using templates

emmby, 2025-03-06:
This ticket is to add a customizable base class to the generate statemachine using standard C++ instead of configs.

The use of the base class is optional. If a user doesn't specify a base class it will use an empty default. 

Specifying a base class will enable the user to easily add methods or variables to their statemachine class, which they can reference from their diagrams.

Example generate class:
```c++
class DefaultBase {};

// Generated state machine
template <typename Base = DefaultBase>
class lightbulb : public Base
{
    ...
}
```

Example usage:
```c++
#include "lightbulb.hpp"

class MyBase {
protected:
    void powerOn() { ... }
    void log( const char* str ) {...}
};

lightbulb<> bulb; // default base
lightbulb<MyBase> bulb2; // custom base

int main()
{
    bulb.start();
    bulb2.start();
}
```

Example diagram:
```plantuml
@startuml
state On
On: enter / {
On:   log("on");
On:   powerOn();
On: }
@enduml

```

adamfk, 2025-03-07:
Can you help explain why you don't want to use toml config?

Would this help: https://github.com/StateSmith/StateSmith/issues/428 ?

emmby, 2025-03-07:
The idea is that a developer who wants to generate C++ from plantuml, for example, can do that using just idiomatic plantuml and C++. No config files, no config strings, no additional knowledge required. Just plug and chug.

More advanced users can still customize things with configs. But for most cases, this change will mean users don't need to know anything about config strings to use statesmith for C++

emmby, 2025-03-07:
Regarding #428 I think it would help, depending on why the user has multiple configs. If the user is trying to do something advanced which requires a config, then it wouldn't help. But if they're just trying to specify a base class, then it would remove the need for multiple configs.

emmby, 2025-03-07:
Looks like this also works in C#, which is nice. Java though will need to use a callback object instead of a supertype because you can't inherit from a generic type parameter in java.

adamfk, 2025-03-07:
"Just plug and chug" 😆 I like it.

Shouldn't the generated `MyStateMachine.hpp` file include the user's `MyBase.hpp` file though? I guess technically not required, but it would make things more user friendly.

Without that, code that uses the state machine will need to remember include both headers, right? Right now, the only way to have the generated state machine include another header file is through the render config.

I don't mind the render config right now because it is fully setup (customized to language) from the ss.cli create command. It's painless.

emmby, 2025-03-07:
hmmm.... that may be a problem...

emmby, 2025-03-07:
Actually it's not bad. This is probably what you meant by not technically required, but you just need to namespace your calls.

eg
```c++
class MyBase {
public:
    void foobar() {}
}
```

Then in the puml you would say
```plantuml
@startuml
[*] --> On
On --> Off
Off --> On

On: enter / {
On:     MyBase::foobar();
On: }
```

Agree that `MyBase::foobar()` is not as friendly as `foobar()`, but it's reasonably intuitive.

emmby, 2025-03-07:
I'll think about it more, but this might be an argument to go back to the original "callback" proposal instead of using inheritance. I think if we just have a callback object instead of inheritance, the user might not need to namespace the method calls. Instead they would just say `callback.foobar()` which would be more intuitive. But my C++ is rusty, will need to take a look.

adamfk, 2025-03-08:
Do you have a small working example with the template idea that you can upload somewhere? I'd like to try playing with it. I think it can work. It's also a very cool feature. Being able to easily change the base type for a state machine could definitely be useful for power users.

My point was just that users of the state machine need to include the file with the base type as well. That's pretty reasonable actually. It's just another step that a user needs to remember.

Template base class:
```cpp
// main.c
#include "MyBase.hpp" // user must remember this!!!
#include "MyStateMachine.hpp"

void blah()
{
  MyStateMachine<MyBase> sm;
}
```

Normal base class:
```cpp
// main.c
#include "MyStateMachine.hpp" // already includes "MyBase.hpp"

void blah()
{
  MyStateMachine sm;
}
```

I think this is a really cool feature (actually 2 features - header only & template base class support)!

I'm not sure that it should be the default behavior though. Templates are generally more confusing for users. I'll need to play with it first though to get a better feel. This is just my initial gut feel.

adamfk, 2025-03-08:
Speaking of base classes, we probably also want to implement this feature https://github.com/StateSmith/StateSmith/issues/443 that allows custom c++ constructors. Required if base class has a constructor with parameters.

emmby, 2025-03-08:
> I'm not sure that it should be the default behavior though. Templates are generally more confusing for users. I'll need to play with it first though to get a better feel. This is just my initial gut feel.

Ideally someone who knows plantuml and c++ should be able to start using statesmith without reading any docs at all. We're actually pretty close. Someone who knows C++ has used them as part of the standard library and so they'll be familiar, even if they may not be an expert. Give them a diagram and plop them in front of ss.cli, and they'd be able to get started right away. That's no longer possible if we require a config language, even though the config may be dead simple.

I'll work on a couple of examples today.

emmby, 2025-03-08:
inheritance and composition examples added to the PR