StateSmith/StateSmith #457: simulator: change the renderer layout to elk

Risele, 2025-04-08:
The mermaid default renderer for state machines is "dagre", which generate hard-to-follow diagrams.

[image]

This example is build with mermaid https://mermaid.live/ and at my PC the original diagram looks much more worse.

The alternative rendered, "elk" provides much more reliable view:

[image]

According to the mermaid API, it could be able to configure via "mermaid.initialize" call.

The layout "elk" have a lot of nice options to customize, including merging the edges , that could be set via TOML (or other way): https://mermaid.js.org/config/schema-docs/config-properties-elk.html

Also, it could be nice to hide the content of the states in the simulator view:

[image]

Risele, 2025-04-09:
It may be a bit more complicated than it looks.

I have tried to manually put the config to the diagram description as [manual says](https://mermaid.js.org/intro/syntax-reference.html#customizing-elk-layout) by placing to the diagrams smth like this, but it takes no effect on layout.
```
---
config:
  layout: elk
  elk:
    mergeEdges: true
    nodePlacementStrategy: LINEAR_SEGMENTS
---
```

I also tried to apply config via `mermaid.initialize` call, but also failed. May be I was not fulfilling it properly, no idea.

adamfk, 2025-04-09:
I remember trying to use other layout engines, but I couldn't get it to work either. 

Any help here would be greatly appreciated!

Risele, 2025-04-10:
> I remember trying to use other layout engines, but I couldn't get it to work either.
> 
> Any help here would be greatly appreciated!

Solution found! 
https://github.com/mermaid-js/mermaid/issues/5969#issuecomment-2645923926 

Attached the very minimum example (all scripting removed)
[minimum.html.txt](https://github.com/user-attachments/files/19685991/minimum.html.txt)

and directly under CLI + manual pasting what's required.
[fromCLI.html.txt](https://github.com/user-attachments/files/19685992/fromCLI.html.txt)

P.S.: also note, than for ROOT state it's fine to use special state [*]: 
https://mermaid.js.org/syntax/stateDiagram.html#start-and-end

adamfk, 2025-04-10:
Thanks!!! This is great! 

I'll start adding this to StateSmith soon. I'm thinking we add a new toml setting for this.

This will allow people to try it out and give us feedback. When we are satisfied, we will make elk the default rendering.

adamfk, 2025-04-14:
Very cool! We can add all of these options to the toml config section, but I also wonder if it would make sense for the simulator User Interface to expose these layout settings to the user? Thoughts?

adamfk, 2025-04-24:
Check this out @emmby. Very cool :)

adamfk, 2026-02-28:
Hi @Risele, I've added this feature in latest alpha CLI release to hide state contents: https://github.com/StateSmith/StateSmith/issues/525

I've done a lot of work on the simulator lately.

I'd like to add Elk support while the simulator code is fresh in my head.

Thanks again for figuring out how to add Elk support :)

I've noticed that Elk avoids a number of rending bugs in default mermaid rendering.

# Issue https://github.com/StateSmith/StateSmith/issues/324
simulator - mermaid.js can't show composite state non-transition behaviors #324

This one trips up users quite a bit. I'd love to fix.

Fix with Elk
[image]

There is an issue in mermaid though in that `&` characters can't be in composite names.

It can also leave a ton of whitespace.

[image]
 

# Issue https://github.com/StateSmith/StateSmith/issues/512
simulator - mermaid bug with parent self transitions #512

No need for workaround with Elk.

[image]

# Issue https://github.com/StateSmith/StateSmith/issues/306
simulator - mermaid.js can't show transitions between parent and child (either direction)

This works fin in Elk :)

[image]